rcmolina/MaxDuino #1: Port to ESP8266 or Esp32

lamerjack, 2019-06-20:
Not exactly an issue:
Somebody already try to port to ESP8266 or ESP32?
Or is not possible?
Main issue is that TimerONE is not supported on ESP.
Esp8266 has similar functions in ticker.h that i hope that work but at the moment i have crashes due to not enough safe GPIO.
ESP32 unfortunately ha not something similar to timerOne.
Will be insterest to use an ESP to have less noises with OLED screens (both works at 3.3v)
And is much faster so maybe will be possible play Antonio Villena cargando leche files.

rcmolina, 2019-06-20:
@lamerjack,

I took a look, but always got dark ...

http://www.esp8266.com/viewtopic.php?f=33&t=2294

Resolution I've seen for other libraries (Ticker and simpletimer)  are not
what we need, timerone works with microseconds....

Not sure if a possible way to redesign to avoid interruptions and
controlling clock cicles from main loop

https://stackoverflow.com/questions/42783027/arduino-esp8266-using-interrupts-to-take-samples

o, got this answered myself.

For the time resolution: you can get a much more exact time by using ESP
.getCycleCount() which counts processor cycles - on my 80 MHz NodeMCU board
that's 12.5 ns per cycle or 80 cycles per microsecond. I should have
mentioned that in the first part.

Interrupts: this is something I misunderstood. Now I solved it by having
the main function wait in a loop until a timeout is reached (set to 1
millisecond, normal expected times are in the 1-100 microseconds range), or
until a global variable is set by the interrupt function. So now I'm
measuring in the 12.5 nanosecond resolution!

One thing that's missing from this sketch is a correction for program time
taken to deal with the timing: the time it takes from dropping the value on
the EC pin to starting to count, and the time it takes from receiving the
interrupt to stopping the count. If that overhead is say 100 cycles, that
would be 1.25 microseconds, which is well within my measurement time.

Hope this helps.

-
Rafael

rcmolina, 2019-07-23:
More info yo check:
https://www.google.es/amp/s/techtutorialsx.com/2017/10/07/esp32-arduino-ti

rcmolina, 2021-02-19:
Email ne if further comments o tests needed.

closed 2021-02-19 by rcmolina