rcmolina/MaxDuino #6: Support Acorn BBC Micro HQ UEf files

PeterBBCUser, 2020-04-23:
Currently does not support HQ (High Quality) UEF files which uses extended chunks see [http://electrem.emuunlim.com/UEFSpecs.htm](url) Chunk &0104 Chunk &0111 etc.

rcmolina, 2020-04-23:
@PeterBBCUser , coud you please clarify your configuration and your problem? Send me your uef file to rcmolina@gmail.com and I will check it !!

Chunks 104 and 111 are well supported. Check my videos:
https://www.youtube.com/watch?v=3ZQr8Xdor-w&t=7s
https://www.youtube.com/watch?v=aPxpDbPDWVE

Regarding protected files, just notice parity is not well preserved in uef files and must be inverted in menu options(TSXCzxpUEFSW). This is automagically done in emulators, but here you need to configure it.

PeterBBCUser, 2020-04-24:
ahh okay maybe that's the reason it does not work, will check and see....

PeterBBCUser, 2020-04-24:
Thanks just changed that setting and it now works, was not aware of it.

closed 2020-04-24 by PeterBBCUser