rcmolina/MaxDuino #7: support for BBC chunk &0117 support 1200/300 baud

PeterBBCUser, 2020-06-11:
Chunk &0117 - data encoding format change
--
This chunk contains a 16 bit unsigned value and is used to select the new   data encoding format. Upon opening a UEF, the "1200 baud" data   encoding format described below should be adopted.    If the value is 300 then the "300 baud" data encoding format is   selected. From henceforth a '0' bit should be encoded as 4 cycles at the base   frequency and a '1' bit as 8 cycles at twice the base frequency.    If the value is 1200 then the "1200 baud" data encoding format   is selected. From henceforth a '0' bit should be encoded as 1 cycle at the   base frequency and a '1' bit as 2 cycles at twice the base frequency.    All other values have undefined effect.

rcmolina, 2020-06-12:
Thanks guy, I know where to read block specification. In stardot you can read there is only one backup that deals with 300 bauds (Billiard) I woud like to understand why to waste time with this. I think an easy response shall move me, but be more friendly before

PeterBBCUser, 2020-06-12:
its not the only program that uses it Doctor Who and the mines of terror uses it extensively for its copy protection and Apple Pie-Visions also makes use of 300 baud for its copy protection.
But I can only ask but if you think its not worth it then fine, just though it was not a big thing to include looking at the code from playUEF https://github.com/8bitkick/PlayUEF/blob/c53feee72f06232bc19cccf6d19efbf3d67feb78/uef2wave.js#L257

rcmolina, 2020-06-12:
Let's make a deal .. send me your files to rcmolina@gmail.com and I'll look at it

rcmolina, 2020-06-12:
Uploaded a beta for testing, only for impatients

PeterBBCUser, 2020-06-12:
lol only one personality just different names for different sites :)
And thanks for having a look at adding the feature to make it more complete :)

closed 2020-06-13 by rcmolina