rcmolina/MaxDuino #8: Automatically pauses on first play

StormTrooper, 2020-10-11:
I've build 2 PCB's with a Atmel 328p and am having an issue when I press the play button it plays but then a millisecond later it pauses. (Almost like a switch debouce but this is not the case)

After I press the play button again I can pause/play without experiencing the issue again. 

At first I thought it was a debounce issue so increased the debounce value to 100ms which had no effect. 
Also as this only happens on first play its  most likely not a debounce issue.

Any ideas as to what the problem is?  I've build 2 boards both with the same issue so don't think its a hardware issue.  

Everything else works fine.

rcmolina, 2020-10-11:
@StormTrooper , configuration has motor control activated by dedaulr, you should enter in the menu and turn it to off. Some computers lile msx have a 2.5 REM jack fir controlling the tape player.

If you have an oled screen it's showed at the centern in the bottom status line.

Hope this solves the issue

merlinkv, 2020-10-11:
Hello

This is because you have the “Motor Control” set to ON. You can change it in Menu options.

Hope this help

StormTrooper, 2020-10-11:
Perfect, thanks for the quick response. That did the trick.

closed 2020-10-11 by StormTrooper