rcmolina/MaxDuino #9: Screen will not rotate.

M0TGN, 2021-02-18:
This worked for my 1306 display :

Updated Display.ino to include this on line 357 - 364

#endif
#ifdef UNROTATE
   
        //sendcommand(0xA0 | 0x1);      //SEGREMAP   //Rotate screen 180 deg 0xA1
        sendcommand(0xA0);
    
        //sendcommand(0xC8);            //COMSCANDEC  Rotate screen 180 Deg /* c0: scan dir normal, c8: reverse */
        sendcommand(0xC0);

Added in userconfig.h on line 56:

#define UNROTATE 1

edited Line 55 :
//#define OLED_ROTATE180  1

rcmolina, 2021-02-18:
Tested on my oled1306, don't see any problem. You are using old code. Commenting/uncommenting the line do the trick.

//#define OLED_ROTATE180

rcmolina, 2021-02-18:
No more UNROTATE

M0TGN, 2021-02-18:
Hi - yeah downloaded the code this AM V1.66 - i uncommented the #define OLED_ROTATE180 line, it made no difference, maybe a crap OLED i have?

rcmolina, 2021-02-18:
If you check code at Display.ino, any difference? here it is:

#ifdef OLED_ROTATE180    
    sendcommand(0xA0 | 0x1);      //SEGREMAP   //Rotate screen 180 deg 0xA1
    //sendcommand(0xA0);
    
    sendcommand(0xC8);            //COMSCANDEC  Rotate screen 180 Deg /* c0: scan dir normal, c8: reverse */
    //sendcommand(0xC0);
//#endif
//#ifdef UNROTATE
#else    
    //sendcommand(0xA0 | 0x1);      //SEGREMAP   //Rotate screen 180 deg 0xA1
    sendcommand(0xA0);
    
    //sendcommand(0xC8);            //COMSCANDEC  Rotate screen 180 Deg /* c0: scan dir normal, c8: reverse */
    sendcommand(0xC0);
#endif

rcmolina, 2021-02-18:
Just in your userconfig.h

rcmolina, 2021-02-18:
Unrotate is the same for //#define OLED_ROTATE180 i.e commented, meaning:
sendcommand(0xA0);
sendcommand(0xC0);

rcmolina, 2021-02-18:
Email me at rcmolina@gnail.com if you need more help so we can close this. I can share you a video so you can watch it..

closed 2021-02-19 by rcmolina