rcmolina/MaxDuino #10: 'class SdFile' has no member named 'cwd'

jduraes, 2021-07-19:
Hello there,

I am getting quite a few errors of this type:

"MaxDuino_v1.67:2023:9: error: 'class SdFile' has no member named 'cwd'" - it happens on quite a few places.

I'm using v1.67 with Arduino IDE v1.8.13, and have already tried several versions of the library SDFat (2.0.7, 2.0.5 and even went as far back as 1.1.4. 

Any ideas on how to resolve this please?

Thanks.

rcmolina, 2021-07-19:
@jduraes, please stay at sdfat 1.1.0
Find attached libraries for the project:
https://github.com/rcmolina/MaxDuino_BETA/tree/master/Troubleshooting/libraries

jduraes, 2021-07-19:
v1.1.0 did the trick - thank you, and for the prompt response as well :-)

Separate question: "Sketch uses 30210 bytes (93%) of program storage space. Maximum is 32256 bytes. Global variables use 1778 bytes (86%) of dynamic memory, leaving 270 bytes for local variables."

Is that ok?

rcmolina, 2021-07-19:
@jduraes, you are using a more recent iD version (my version is 1.8.4). results and optimization depends much on this. This is what I get for old Arduino Nano with usb programming:
Archiving built core (caching) in: C:\Users\Rafael\AppData\Local\Temp\arduino_cache_164344\core\core_arduino_avr_pro_cpu_16MHzatmega328_f8d539391cd9a049a88b673cece52bfa.a
El Sketch usa 30568 bytes (99%) del espacio de almacenamiento de programa. El máximo es 30720 bytes.
Las variables Globales usan 1749 bytes (85%) de la memoria dinámica, dejando 299 bytes para las variables locales. El máximo es 2048 bytes.
Poca memoria disponible, se pueden producir problemas de estabilidad.

rcmolina, 2021-07-19:
Prevoius report with OLED diplay. Are you using LCD?

rcmolina, 2021-07-19:
For LCD Display:
El Sketch usa 29464 bytes (95%) del espacio de almacenamiento de programa. El máximo es 30720 bytes.
Las variables Globales usan 1753 bytes (85%) de la memoria dinámica, dejando 295 bytes para las variables locales. El máximo es 2048 bytes.
Poca memoria disponible, se pueden producir problemas de estabilidad.

closed 2021-07-19 by rcmolina

jduraes, 2021-07-19:
So you get a similar message and it works for you fine, yes? 

I am using OLED as well.

[https://photos.app.goo.gl/XX9qeW9o4kQamx8Z6](url)

rcmolina, 2021-07-19:
Of course, this is only a warning. You can experiment some hanging if you are pushing to the limits, but works.

jduraes, 2021-07-19:
Cool, thank you. I don't get anything on the OLED; do I need to configure something in the source code, or should I recheck my wiring? Already checked it once :-)

rcmolina, 2021-07-19:
My friend, you could check this easily with OLED examples in the IDE :)
As you know, Maxduino config for arduino nano is in file userconfig.h, not much to change:

//#define LCD_I2C_ADDR    0x27        // Set the i2c address of your 1602LCD usually 0x27
//#define LCD_I2C_ADDR    0x3f        // Set the i2c address of your 1602LCD usually 0x3f
//#define LCDSCREEN16x2             // Set if you are using a 1602 LCD screen

//#define OLED_SETCONTRAS   0xcf      // Override default value inside Diplay.ino, bigger to increase output current per segment
#define OLED_ROTATE180
#define OLED_address   0x3C           //0x3C or 0x3D
#define OLED1306                      // Set if you are using OLED 1306 display
    //#define OLED1306_128_64         // 128x64 resolution with 8 rows
    //#define OLED1106_1_3            // Use this line as well if you have a 1.3" OLED screen
    #define video64text32

jduraes, 2021-07-19:
You are of course correct :-) It's too hot and I can't think properly...

Thank you for all your help.