rcmolina/MaxDuino #11: Artifacts on OLED, not going away with a reset

raduc, 2022-02-19:
I recently started to see these artifacts on screen on a Maxduino Slim (see the right side), which don't go away with a reset. I tried old firmwares and different user configs but they still show up. Can it be that the reset doesn't happen correctly?
I replaced the OLED as well but it still happens with a new one (SSD1106 as well).
Any suggestions on what may happen?
The only similar issue I've seen is this: https://forum.arduino.cc/t/how-to-get-rid-of-the-white-line-on-the-right-side-of-my-oled/492264
I wonder if anyone else has seen similar issues.
[image]

Thanks.

rcmolina, 2022-02-20:
@raduc, as far I can see, your oled screen is not well configured. (ManuFerhi Slim version with OLED 1,3"?)
If you are using optiboot (max 32256 bytes), use option 2. If not, and using old bootloader, use option 7 but uncomment
 #define OLED1106_1_3:

#define OLED1306                      // Set if you are using OLED 1306 display
    #define OLED1306_128_64         // 128x64 resolution with 8 rows
   #define OLED1106_1_3            // Use this line as well if you have a 1.3" OLED screen <- THIS ONE, REMOVE INITIAL SLASHES
    //#define video64text32

raduc, 2022-02-20:
I figured out I was missing the OLED1106_1_3 part and I was able to use up to V1.71. However, for newer firmwares, if I enable   CONFIGFILE = 2 the sketch will be too large. How do I enable optiboot, do I need to re-flash the bootloader on Maxduino?
And yes, it is a ManuFerhi Slim version with OLED 1,3" and so far I used Arduino Nano/old bootloader in Arduino IDE.
Thanks.

rcmolina, 2022-02-20:
@raduc, watch my videos at: https://m.youtube.com/channel/UC0L03I8mb4CMytW7MuhF--g/videos

Te one with optiboot bootloader update is here: https://m.youtube.com/watch?v=Bo8B5Gl8UWg

closed 2022-02-20 by rcmolina

raduc, 2022-02-20:
Ok, I flashed 1.70 with optiboot as per the video instructions and 1.70 works fine now, but I am still unable to flash 1.76 with option 2 config, it still complains of sketch too big for Nano.

raduc, 2022-02-20:
BTW, it loooks like it is working with option 7 with #define OLED1106_1_3 uncommented, but that one is still using old bootloader so it would still be good to e able to use option 2 with optiboot. What other differences are between 2 and 7, am I missing anything by using option 7 with optiboot flashed Maxduino?

rcmolina, 2022-02-20:
@raduc, note this is not a Maxduino problem, but an Arduino IDE. You can install optiboot package, and use new procs on menu with updated capacity. Also, you can go to Boards and properties and create manually this max size value and comm protocol.
As an alternative, while you get knowledge, I can upload the hex with optiboot sou you can flash it.

rcmolina, 2022-02-20:
@raduc, of course we have some feature differences. If you look at userconfig.h, where you select your option, there i a summary o features for every option. But the more secure way y to compare both configs and watch commented/uncommented lines.

More space, more features like handling more type if blocks on oled screen :)

rcmolina, 2022-02-20:
@raduc, Arduino UNO board comes with optiboot,  try to configure if on your IDE, good point to start :)

raduc, 2022-02-21:
Thanks, that was very helpful. I was able to upload config 2 now with either default Uno board or with a modified Nano board (I set max size to 32256 (instead of 30720), just like the Uno entry.