rcmolina/MaxDuino #14: (Maybe a feature request) AY Files are always loaded using a "ZXAYFile" filename, rather than showing filename from sdcard

stripwax, 2022-06-07:
Would be quite easy to take the filename, reduce to 10 characters (pad with space if necessary, and remove characters outside of the 32-127 range), and send that as part of the writeHeader2() function

stripwax, 2022-06-07:
Please consider https://github.com/stripwax/MaxDuino/commit/8c8b6893d957600300c38187ab4b439233886c2f

rcmolina, 2022-06-11:
> Please consider [stripwax@8c8b689](https://github.com/stripwax/MaxDuino/commit/8c8b6893d957600300c38187ab4b439233886c2f) as a solution (apologies I didn't make this a pull request, I can do that if you prefer)

Yes please

stripwax, 2022-06-11:
done (#18)

closed 2022-06-11 by stripwax

rcmolina, 2022-10-11:
Thanks David,

Many of these error are in fact attempts to get more space, and not well
tested.

No user report this, and not observed by myself.

Rafa