rcmolina/MaxDuino #37: 8n1 8e1 parity MakeUEV 2.4

PeterBBCUser, 2023-04-08:
There has been an update of MakeUEF to version 2.4 there is a long standing bug where the parity 8n1 8e1 where swapped over, but has now been fixed but this fix breaks MaxDuino and other similar units for any UEF file that uses parity 8n1 8e1 in newly created UEF files.

rcmolina, 2023-04-08:
Hi Peter, yes I kow that. This question arises in my head years ago, and what should be the best solution cause therevare may files bad preserved out there... So, my decision then was to set as default the old parity preservation, and the posibility to chabge this manually: just enter in the menu and change tsxcontrol/zax polarity/SWITCH CONTROL to swap parity.

Is this ok for you, or you want to set default behaviour?

Thanks for your interest on this.

Rafa

stripwax, 2023-04-08:
Is there any way to identify a "good" file from a "bad" file?

rcmolina, 2023-04-08:
@David, of course. There is a header to deal with makeuef version number... But I know only 7 files with the parity protection.. @Peter, if you submit your file we can check it and think about this. If not not working with switching, not suere because you say no parity and not odd parity, this will be chedked and changed. 

Rafa
-------
http://www.stairwaytohell.com/bbc/archive/tapeimages/BubbleBus/Starquake_B.hq.zip
--
UEF File!
MakeUEF V1.8.^UEFed by Fraser Ross. Cassette player was Acorn ANF03. Created for www.acornpreservation.org.The input format was CSW.� AL/

rcmolina, 2023-04-08:
More info here:

https://stardot.org.uk/forums/viewtopic.php?p=389798#p389798

PeterBBCUser, 2023-04-08:
Ah okay I did not realize the setting "TSXCzxpUEFSW" (I assume this is the setting)  was to change that, I will give it a go.

@stripwax: The only way to tell I guess is to look at the header of the uncompressed UEF to see if its made with MakeUEF 2.4 (the now fixed version) and MakeUEF 2.3 that creates the switched parity if 8n1 8e1 are used which is only really used in copy protected titles.

closed 2023-04-08 by PeterBBCUser