rcmolina/MaxDuino #38: Version 1.76 causes Garbage on the display

PeterBBCUser, 2023-04-08:
I tried to upgrade my unit to the latest Version 1.76 (from version 1.56) but when I press the stop button I get Garbage on the display (see picture).
[image]

rcmolina, 2023-04-08:
Are you using this version? : https://github.com/rcmolina/MaxDuino_BETA/tree/master/Troubleshooting/MaxDuino_v1.76

Can you submit me to rcmolina@gmail.com your user config file if changed?

Rafa. 

Ps. Remember this is not the last, many more optimization from then..

PeterBBCUser, 2023-04-08:
sorry its when I press the menu button it does the odd things.
it was the latest files I downloaded from https://github.com/rcmolina/MaxDuino/archive/refs/heads/master.zip 
"Select config file 3" I am not sure how you do that as that's one of the issues I am having I am not sure how to get it to change config files as I see there are multiple files but i need the one for a 16*2 lcd addr 0x27 so to get it working i have been removing the userconfig.h file and renaming the userconfig0.h to userconfig.h (where i have rem'ed out some of the lines) or else I don't get any display, attached is a copy of the file.
[userconfig.zip](https://github.com/rcmolina/MaxDuino/files/11184246/userconfig.zip)

rcmolina, 2023-04-08:
Hi Peter,

It's easy. Now I've added a frontend and have predefined configs fir
multiple devices.

You need only to change the first line in:

https://github.com/rcmolina/MaxDuino_BETA/blob/master/Troubleshooting/MaxDuino_v1.76/userconfig.h,
from 7 to 3 to import configfile with lcd

#define CONFIGFILE 3

PeterBBCUser, 2023-04-08:
Hi Rafa,

Ah okay I must have missed that in the documentation how to set up or use the pre defined configs, only I see 3 is for OPTIBOOT BOOTLOADER I am still using the old bootloader method and wonder if thats what is giving problems.

stripwax, 2023-04-08:
I've changed the menu code recently, and haven't tested on 16x2 lcd - maybe
that's related - does the problem occur with an older version? (it's a bit
fiddly but you can just download any older version directly from github
e.g. try this one
https://github.com/rcmolina/MaxDuino/tree/43182de759b8abe2c6cacc454dd8e4d0b334a24d
and click on the green Code button then Download)
We'll make versions more clear soon. right now, we haven't formally updated
the version number since 1.76 (the last update in the readme says 1.76 - we
should actually use Releases on github instead)

stripwax, 2023-04-08:
Oh.. I think I can see what the bug might be.  If you edit menu.ino and
change

printtextF(MENU_ITEMS[menuItem], lineaxy);
to

printtextF((char *)(pgm_read_ptr(&(MENU_ITEMS[menuItem]))), lineaxy);

does that help at all?

PeterBBCUser, 2023-04-09:
thank you for your reply and yes that fix does sort out the Menu problem it now displays the settings as expected.

closed 2023-04-09 by PeterBBCUser

stripwax, 2023-04-09:
@PeterBBCUser yep, apologies for the problem and I've created a pull request for the fix

commit da2bc358 (2023-04-09): fix #38 - correct earlier PROGMEM change for menu items