rcmolina/MaxDuino #44: MaxDuino doesn't show "Stopped" when you stop a file.

stripwax, 2023-04-21:
This is probably a pretty old bug.  When you play a file, and then hit Stop button, MaxDuino never displays "Stopped".
From the code, it looks like it's supposed to, and maybe it used to in the past, but it doesn't do that now.  This is because "Stopped" is only displayed by the stopFile function if "start==1", and start will always be zero at that point because stopFile already called TZXStop which sets start to 0.
Furthermore, TZXStop not only sets start to zero, but also calls seekFile(), which ensures that the filesize is redisplayed on the top line (instead of Playing).

So, this seems intentional in some way.  Personally, I would prefer to see the filesize (as current behaviour) instead of "Stopped", after stopping a file (it would be obvious it has stopped).  But it also means P8544 users never see the STOP icon.  So it must be a bug.

Question is, what is the fix that people want?  Proposal: keep the current behaviour (show filesize, not "Stopped", when stopping a file), but fix it so that P8544 users see the icon.  And remove the printtextF(PSTR("Stopped")) (to save a few bytes since it never gets displayed)

stripwax, 2023-04-21:
Comments about P8544 here are also related to #45 .  maybe P8544 isn't supported and code can be removed entirely :-)

rcmolina, 2023-05-07:
My personal preference is not to show stop, I thik it' no a useful info, you know it's stopped because you see the file size again.

The code for Nokia monochrome screen .. , yes it can be removed. Ath very beginning I thought it was clearly a cheap alternative to lcd, but not spread. So also unuseful now.

closed 2023-05-07 by rcmolina