rcmolina/MaxDuino #52: Adaptar MaxDuino al ESP32 Audio Kit

hash6iron, 2023-05-08:
Buenas tardes Rafa,

Adquirí en Alliexpress hace unas semanas un kit de audio con ESP32, equipado con 8 pulsadores, una salida de audio stereo, tarjeta SD, entre otras mas cosas por menos de 15€. Me preguntaba como de complicado sería adaptar maxduino, para compilarlo en esta placa, ya que tan solo habría que conectar una pantalla LCD, y sería genial para montar MaxDuino rapidamente.

Espero tus comentarios.

Muchas gracias, y mi agradecimiento por tu excelente trabajo.
Un cordial saludo.

stripwax, 2023-05-08:
I had a quick look and it's unclear if the ESP-ADF is actually supported under Arduino (and MaxDuino is an Arduino project).  Assuming it is not supported, this would be a very hard project.

hash6iron, 2023-05-08:
Hello.

This audio kit is supported thanks to this library,

https://github.com/pschatzmann/arduino-audio-tools

My board is proggramed with "ESP32 Dev Module" board, under new Arduino IDE.

Thanks.

stripwax, 2024-09-23:
What I would suggest, is that you take a look at the sourcecode of MaxDuino and assess for yourself if this is indeed an easy change or a hard change.  I added the support for the ESP32 (specifically the Seeed Xiao ESP32C3), output is mono on a single pin, buttons are read via ADC on a single pin, SD card and Display support were provided already by the MaxDuino codebase from the standard devices that were already supported by the codebase.
You might indeed find it's quite easy to modify and add support, and I'm sure we would be interested in your patches.

stripwax, 2024-10-25:
Closing as no comments or responses received