rcmolina/MaxDuino #55: TRS-80 CAS files

DFL1973, 2023-06-22:
Is it possible to support TRS-80 CAS files? Not TRS Color ones, the TRS-80 seems to be 250/500/1500 bauds.

Examples https://www.classic-computers.org.nz/system-80/software_cassette_archive.htm

rcmolina, 2023-06-22:
I'll take a look, main inconvenient is I don't have rhis machine neither an fpga core for my sidi. If you help me, testing, for sure..

DFL1973, 2023-06-22:
> I'll take a look, main inconvenient is I don't have rhis machine neither an fpga core for my sidi. If you help me, testing, for sure..
> 

I can test in the actual machine for sure, thanks.

I don´t know if it helps, but there is a PC app that plays those CAS. http://knut.one/PlayCAS.htm

> Also, one stupid question: Did you explore the generic procedure convertig wav to tzx id15, this is an easy solution for sure.
> 
> Did you check this?
> 
> https://github.com/rcmolina/ColourGenie

I didn´t even know about this conversion. I can try this, thanks!

stripwax, 2024-09-02:
I had a quick look, seems like it could be an interesting addition, but is there any doc for the TRS-80 CAS format and (if) how it differs from the other formats already supported by MaxDuino?  q for @DFL1973

stripwax, 2024-10-25:
@DFL1973  Would you be able to take a look at the question and respond please?
1. Did generic conversion to ID15 work for you?  (Noting that we've got significant performance improvements for ID15 in the latest build, so try that if you haven't)
2. Do you know of any documentation?

Otherwise we'll close this item as not supported.  Thanks!

Snocksman, 2024-10-25:
Hi !

A few days ago I actually tried to load a few programs onto my TRS-80 with my Maxduino and found that it didn't work. There is a small Windows tool that can be used to play CAS files for the TRS-80. After listening to the two sound outputs (Maxduino and this tool), it quickly became clear to me that Maxduino was playing the files way too fast. Some more information, the Windows Playback Tool and also .cas files for the TRS-80 can be found here: [https://www.classic-computers.org.nz/system-80/software_cassette_archive.htm](url)

It would be really cool if you could add support for TRS-80 .cas Files !

stripwax, 2024-10-25:
Cool, I'll check it out

Snocksman, 2024-10-25:
If you have something I can test with my Maxduino, I'll be happy to do it.

stripwax, 2024-10-25:
Some info and guesswork - correct me if I'm wrong please!
Doing some digging, there are multiple formats called 'CAS'.
One is the "Atari" style CAS format, most likely the one originated by
Ernest R. Schreurs around 1997.
This is different to the "MSX" style CAS format, and has never been
supported by MaxDuino (and probably nor by the other *duino devices). It's
a different format.

Seems like the TRS-80 CAS format is based on the Atari CAS format; there
are multiple different interpretations of this format to handle different
baud rates and turbo mode; there are extensions documented here for (yet
another) format :
https://a8cas.sourceforge.net/format-cas.html#:~:text=CAS%20is%20a%20binary%20file,tapes%20with%20copy%2Dprotection%20measures
.

I'll play around and bit and see what I can figure out.  If you have any
more pointers, that would be great;  it will be slow progress without
detailed binary/protocol formats.

Snocksman, 2024-10-26:
Maybe this Link can bei interesting, too... Here you can load .CAS Files and browse through the Data of the File...

[https://www.my-trs-80.com/cassette/](url)

rickyelqasem, 2024-12-23:
I just saw this, but I had a mate test out  TRS-80 coco CAS files on Maxduino a few years back, and they worked. I don't know if they are different, but I doubt it.

rickyelqasem, 2024-12-23:
Not sure if the non color TRS CAS files are different from the COCO ones but here is a picture my mate sent me with games working > https://www.youmakerobots.com/299-large_default/cocouitape-trs-80-sd-solution.jpg

Snocksman, 2026-03-18:
I haven't checked in here for a while, but yesterday I held the TRS-80 in my hands again and thought, what a shame it didn't work with Maxduino... @rickyelqasem as far as I know, there is a difference between the CoCo and the black and white TRS-80, like the one I have. The CoCo is also compatible with the Dragon32, which the TRS-80 apparently isn't.

stripwax, 2026-03-18:
> OK, that's really a shame...

I think we can continue the conversation in a different issue in one of the forks, probably ricky's:
(@rickyelqasem  can you turn on Issues in your repo fork?). The outcome of that might then be a PR that Rafa can review

https://github.com/rickyelqasem/MaxDuino