rcmolina/MaxDuino #62: stm32 bases maxduino

lotharek1977, 2024-10-24:
Hello, i have designed small stm32 based maxduion compatible device.  link is here: https://lotharek.pl/productdetail.php?id=409 

can i please send a proto board to You to verify maxduino software is fully functional. my test with zx spectrum show no issues. Please, contact me to lotharek@lotharek.pl :-)

stripwax, 2024-10-25:
I don't think that's a service that we can provide for you.
Sorry!
In particular, my understanding is that the STM32 functionality is not currently supported and has no active maintainer.
Perhaps you would be interested in doing the testing yourself, with your devices, and providing any feedback or patches to this project?

However, I'm keen to understand the advantage of the STM32-based device here.  How does this compare against the other devices, including things like the SAMD21 or ESP32 which is now actively maintained (by me 😄  )

lotharek1977, 2024-10-25:
Hello. stm32 actually works very good. arduino and usb /maple programming works ok.  so far i have found one issue - cannot play next file withot pressing STOP after previous play is finished.  please, lool at video here:  https://lotharek.DSCloud.biz:5001/sharing/nHyG4q9Si  / my synology server link/. best regards

stripwax, 2024-10-25:
Can you submit a bug report or, ideally, a patch for the issue(s) found please.  Also, please take a look at the questions from my earlier response.  Thanks!

stripwax, 2024-10-25:
Two different issues please.
1.  Your request for me to do your product testing for you  :-)
2. Your bug report
(each different bug, should have a separate report in a separate issue)

That way we can track individual issues and mark one as fixed when it is fixed.

Having one issue with already two different things in it = not helpful for me.  Thanks!

lotharek1977, 2024-10-25:
OK. thnak You :-)

closed 2024-10-25 by lotharek1977