rcmolina/MaxDuino #65: STm32 build play/stop issue

lotharek1977, 2024-10-25:
Hello, i have designed small stm32 based maxduino compatible device. 

link is here: https://lotharek.pl/productdetail.php?id=409

- stm32 build  works very good in general.. 
- arduino IDE 1.9.xx (!!) and usb /maple programming works ok

so far i have found one issue - cannot play next file withot pressing STOP after previous play is finished

please, look at video here: https://lotharek.DSCloud.biz:5001/sharing/nHyG4q9Si / my synology server link/

best regards

stripwax, 2024-11-04:
Sorry that I cannot offer more specific advice, but can you try updating to latest arduino IDE, and all latest packages?  There is no reason to build using old tools.

stripwax, 2024-12-10:
You can now also try the latest releases built automatically and downloadable here.  Try those and let us know if the issue is still there or not.

lotharek1977, 2024-12-13:
encounter some problems UTOA related:

C:\Users\Asus\Desktop\MaxDuino\menu.cpp: In function 'void menuMode()':
C:\Users\Asus\Desktop\MaxDuino\menu.cpp:154:47: error: 'utoa' was not declared in this scope
               utoa(baudrate, (char *)input, 10);
                                               ^
C:\Users\Asus\Desktop\MaxDuino\CounterPercent.cpp: In function 'void lcdTime()':
C:\Users\Asus\Desktop\MaxDuino\CounterPercent.cpp:52:43: error: 'utoa' was not declared in this scope
           utoa(lcdsegs%10,(char *)input,10);
                                           ^
C:\Users\Asus\Desktop\MaxDuino\CounterPercent.cpp:57:56: error: 'utoa' was not declared in this scope
           utoa((lcdsegs % CNTRBASE)/10,(char *)input,10);
                                                        ^
C:\Users\Asus\Desktop\MaxDuino\CounterPercent.cpp:64:67: error: 'utoa' was not declared in this scope
           utoa((lcdsegs % (CNTRBASE*10))/CNTRBASE,(char *)input,10);
                                                                   ^

rcmolina, 2024-12-13:
@lotharek1977, this is not specific problem for STM32, and compiling well for me with other microcontrollers setup.

This C function belongs to stdlib, so you can try:
1) check if changing to itoa compiles well.
2) include manually <stdlib.h>
https://github.com/leaningtech/cheerp-newlib/blob/master/newlib/libc/stdlib/utoa.c
3) Check your environment arduino ide/platformio versions and upgrade as needed.

lotharek1977, 2024-12-13:
**check if changing to itoa compiles wel**

No. the same problem.
                                                                 ^
Previous firmwares, nevertheless also include utoa string compiles. 
Arduino - today latest version. 

[image]

rcmolina, 2024-12-13:
Thanks for the additional info,  you are compiling with arduino IDE 2.3.4.

I've checked with Arduino IDE 1.8.4, works well with Arduino Pro mini, but changing board to Maple mini and compiling throw this errors,  So verified. Need further investigation, we will check this.

lotharek1977, 2024-12-13:
1.8.19 - compilation/upload OK
2.3.4 - no succes so far...upload stopped working for me at all after upgrade. 

_C:\Users\Asus\AppData\Local\arduino\sketches\7FB9F4D9C1C4B4FD238611705096F128\sketch\CounterPercent.cpp.o: In function `lcdTime()':
C:\Users\Asus\Desktop\MaxDuinoSketch uses 47344 bytes (43%) of program storage space. Maximum is 108000 bytes.
Global variables use 4752 bytes (27%) of dynamic memory, leaving 12248 bytes for local variables. Maximum is 17000 bytes.
/CounterPercent.cpp:57: warning: undefined reference to `utoa(unsigned int, char*, int)'
C:\Users\Asus\Desktop\MaxDuino/CounterPercent.cpp:62: warning: undefined reference to `utoa(unsigned int, char*, int)'
C:\Users\Asus\Desktop\MaxDuino/CounterPercent.cpp:69: warning: undefined reference to `utoa(unsigned int, char*, int)'
C:\Users\Asus\AppData\Local\arduino\sketches\7FB9F4D9C1C4B4FD238611705096F128\sketch\menu.cpp.o: In function `menuMode()':
C:\Users\Asus\Desktop\MaxDuino/menu.cpp:158: warn_ing: undefined reference to `utoa(unsigned int, char*, int)'

will report funcionality within 30 minutes

lotharek1977, 2024-12-13:
effect:

https://lotharek.DSCloud.biz:5001/sharing/Ii74kTqDB

rcmolina, 2024-12-13:
@lotharek1977, thanks for the evidence. As I told you before, I have an old STM32 gadget by Antonio Villena, will update also and share the results in this thread,

lotharek1977, 2024-12-13:
small update: after uploading with 2.3.4 arduino with st -link:

https://lotharek.DSCloud.biz:5001/sharing/IuO8spOTf

rcmolina, 2024-12-13:
Included "compat.h" instead of extern *utoa. Uploaded changed files (if should compile now with Arduino IDE 2.3.4)

stripwax, 2024-12-13:
I'll take a look later, I wonder why it failed to build on Arduino IDE but
succeeded on GitHub/platformio . Maybe I need to tweak something.

stripwax, 2024-12-13:
I compiled fine on Arduino IDE, and I did not receive the warning
message: "undefined
reference to utoa(unsigned int, char*, int)'"

But some codepaths depend on what config settings you have.
So please send your config file, so we can reproduce exactly your problem.

What I suspect is happening is that you have a stale version of the STM32
board package, *OR* that I am using a different/wrong version of the STM32
board package / device.  Maybe that's why I don't see the error, and you do?
Which board/device are you selecting in the IDE?
see screenshot:
[image: image.png]

stripwax, 2024-12-13:
Darn, screenshot didn't work, trying again:
[image]

stripwax, 2024-12-13:
I'm using the SMT32F1 Boards package 2022.9 as you can see:
[image]

stripwax, 2024-12-13:
Oh you must be using LCDSCREEN16x2 , not OLED?  I understand now.  I'll
update the build process accordingly, because this combination is not
built and tested.

stripwax, 2024-12-13:
Oh you must be using LCDSCREEN16x2 , not OLED?  I understand now.  I'll update the build process accordingly, because this combination is not built and tested.
But I even tried building this using the earlier code (without Rafa's change) and I still did not see that error message.
So that's what I want to understand first.  That change might not be needed.  (Since I don't need it and it compiles for me)

By the way i tried to view the video but got this 
[image]

lotharek1977, 2024-12-16:
Gentlemen, 

the same here 
[image]

arduinoIDE 2.3.4, 

upload method stlink, device: generic stm32f103c at the moment

lotharek1977, 2024-12-16:
> Included "compat.h" instead of extern *utoa. Uploaded changed files (if should compile now with Arduino IDE 2.3.4)

compilation OK, upload OK.
testing

stripwax, 2024-12-16:
Sorry I'm behind here, I still cannot reproduce the problem and I'm also still trying to validate the changes that Rafa added.  Those changes didn't seem required for me, and I'm using the code straight from Rafa's master branch at the point of my latest merge commit (2e93065605e97ef12160eec1c9eae44651c648ff)

It compiles for me, so I think we're somehow looking at some parallel problems-
1. a compilation problem
2. a behaviour problem

I want to just address 1 first so that we are all on the same page (as noted, I don't have any compilation problem and I'm using the same code, unless you have some changes?)

Your screenshot has cut off part of the board number so can you confirm please, are you using:

"Generic STM32F103C Series"
or
"Generic STM32F103C6 Series/fake STM32F103C8"

stripwax, 2024-12-16:
For reference, here's my local build output with Platform IO

```
Environment                        Status    Duration
---------------------------------  --------  ------------
seeed_xiao                         SUCCESS   00:00:11.868
Nano328p                           SUCCESS   00:00:06.076
Nano328p_LCD16                     SUCCESS   00:00:05.954
Nano328p_CF0                       SUCCESS   00:00:05.394
Nano328p_CF1                       SUCCESS   00:00:05.264
Nano328p_CF2                       SUCCESS   00:00:06.252
Nano328p_CF3                       SUCCESS   00:00:05.888
Nano328p_CF4                       SUCCESS   00:00:05.200
Nano328p_CF5                       SUCCESS   00:00:05.587
Nano328p_CF6                       SUCCESS   00:00:05.333
Nano328p_CF7                       SUCCESS   00:00:05.361
Nano328p_CF8                       SUCCESS   00:00:05.353
Nano328p_CF9                       SUCCESS   00:00:05.130
Nano328p_New                       SUCCESS   00:00:05.511
megaatmega2560_MegaDuino           SUCCESS   00:00:05.430
ThinaryNanoEvery_MaxduinoUltimate  SUCCESS   00:00:05.829
ESP8266                            SUCCESS   00:00:16.237
ESP32C3                            SUCCESS   00:00:24.904
STM32_MapleMiniDuino               SUCCESS   00:00:09.216
STM32_MapleMiniDuino_CF0           SUCCESS   00:00:08.907
```

Here's my local build output with Arduino IDE, using board "Generic STM32F103C Series"
```
Archiving built core (caching) in: C:\Users\slowd\AppData\Local\arduino\cores\21baac7b6fae8205df4b3340d2cfb791\core.a
Linking everything together...
"C:\\Users\\slowd\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -Os -Wl,--gc-sections --specs=nano.specs -mcpu=cortex-m3 "-TC:\\Users\\slowd\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\variants\\generic_stm32f103c/ld/bootloader_20.ld" "-Wl,-Map,C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95/MaxDuino.ino.map" "-LC:\\Users\\slowd\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2022.9.26\\variants\\generic_stm32f103c/ld" -o "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95/MaxDuino.ino.elf" "-LC:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95" -lm -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\CheckForExt.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\CounterPercent.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\Display.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\LiquidCrystal_I2C_Soft.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\MaxDuino.ino.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\MaxProcessing.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\TimerCounter.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\USBStorage.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\ayplay.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\buffer.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\buttons.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\casProcessing.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\current_settings.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\file_utils.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\i2c.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\isr.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\kansas_4b.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\menu.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\oric.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\pinSetup.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\power.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\processing_state.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\uef.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\utils.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\sketch\\zx8081.cpp.o" "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95\\libraries\\SdFat\\ExFatLib\\ExFa [... cut ...]
Using library EEPROM in folder: C:\Users\slowd\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26\libraries\EEPROM (legacy)
"C:\\Users\\slowd\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-size" -A "C:\\Users\\slowd\\AppData\\Local\\arduino\\sketches\\0D9C3106E5FABABB1FF588803731AE95/MaxDuino.ino.elf"
Sketch uses 46772 bytes (71%) of program storage space. Maximum is 65536 bytes.
Global variables use 4656 bytes (22%) of dynamic memory, leaving 15824 bytes for local variables. Maximum is 20480 bytes.
```

stripwax, 2024-12-16:
Oh - that's interesting - am I using a different version of gcc somehow, because of the Seeeduino package?

```Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++```

What version of g++ shows up in your logs?

lotharek1977, 2024-12-16:
hello

1. generic stm32f103 - not fake ;-)
2.  compilation ok; please advice what shall i show ( i use clean install of aruidno 2.3.4 , sdfat most present lib and for board manager https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
3. funcionality - one block can pbe played at the time and software problem from initial post reappears . 
4.   video comparing OTLA firmware (first part, present release and previous firmware - second part)
 https://lotharek.DSCloud.biz:5001/sharing/KNOFOmDj4

stripwax, 2024-12-16:
Thanks for 1 and 2 (we'll get to 3 and 4 as soon as we can).  Can you provide your config file i.e. userSTM32config.h  (I am assuming you are not using this default config, and have made some changes).  (The reason I am assuming this is because I cannot reproduce your earlier compilation failure, which means what I am testing is not the same as what you are doing, so we need to get there first)

lotharek1977, 2024-12-16:
> Thanks for 1 and 2 (we'll get to 3 and 4 as soon as we can). Can you provide your config file i.e. userSTM32config.h (I am assuming you are not using this default config, and have made some [...]

to be honest - i use default config now of present OTLA firmware. and to be honest even more, past firmware this pull request started, was also default.   meanwhile was playing with diffrent lcds/logos/etc but basic functionality was never touched as device works

stripwax, 2024-12-16:
Ok, that's perfect thanks.  I still don't know why I don't see the same compilation error as you, but I'll take a look at the behaviour.
Can you copy+paste the compilation output from IDE 2.3.4 and add it as a file attachment here please (it will be large, but don't worry about that).

And the issues you're reporting - with buttons, and playback - that hasn't actually changed with the latest OTLA version, you still have just the same problems?  My guess is the STM32 timer code just doesn't work properly, maybe it has been buggy for some time, but I'm not sure who is supposed to be the STM32 platform expert right now since that code was contributed by someone else...   I'll try to help....

lotharek1977, 2024-12-16:
> Ok, that's perfect thanks. I still don't know why I don't see the same compilation error as you, but I'll take a look at the behaviour. Can you copy+paste the compilation output from IDE 2.3.4 and [...]

Last OTLA firmware, mentioned above " Included "compat.h" instead of extern *utoa. Uploaded changed files (if should compile now with Arduino IDE 2.3.4)" gives no errors.  

bahavior of 2 devices:

-  OTLA (not correct loading of the first file - the same computer bahavior - not receivng data)
- previous  firmware  (correct loading of the first file, but cannot load next ZX file after stop/play again) 

  are shown at  video i linked here:  https://lotharek.dscloud.biz:5001/sharing/KNOFOmDj4

stripwax, 2024-12-16:
Please attach the compilation output anyway, many thanks.   Even though there are no errors now with the change from Rafa, I'm still interested in the version of the compiler and other packages, as I was not able to reproduce (see my comment about Seeduino package - I'm assuming you have not got those installed).   I want to understand why you needed Rafa's change and I did not.  I want to get to the point where we are all definitely building the same thing.

On that note, you can also download the prebuilt images from github now  -  but based on your reports, I assume you will see the same problematic behaviours.

stripwax, 2024-12-16:
:-)  Seeeduino is a range of Arduino-compatible devices, I added support for Xiao M0 and ESP32C3 to MaxDuino.  As a result, I have the Seeeduino package installed and, even when compiling for STM32, I can see it is using the compiler version from the Seeeduino package.  (Which seems odd to me, and totally out of my control;   it's very hard to control what Arduino IDE is actually doing when one package seems to overlap with another package).

https://wiki.seeedstudio.com/Seeeduino-XIAO/

stripwax, 2024-12-16:
Oh you sent me a zip of the files.  I was looking for just the compiler output from the Arduino IDE output window.

rcmolina, 2024-12-16:
@lotharek1977 and @stripwax , I'm doing some testing with my old mapple STM32 minduino:
-loading turbo files work well at 5800 bauds.
- No play/stop issue ¿?
- OTLA not working well with zx spectrum, seems slow when playing.
- Other platforms testing cpc, oric, msx in some minutes. I'll edit this post.

stripwax, 2024-12-16:
Thanks @rcmolina , is that also using the same "Generic STM32F103C Series" and userSTM32config file?  If not I want to ensure we have a few varieties defined in platformio.ini so we can keep building the ones people actually use/need

rcmolina, 2024-12-16:
@stripwax , also remeber one important thing: library SoftI2CMaster was not working with STM32 platform, so no acceleration for i2c.

stripwax, 2024-12-16:
I know. I don't think that is important here though or is it?

rcmolina, 2024-12-16:
@stripwax , not tested with new code. But before I had some limit reaching 5800 bauds.
Easy to check again with nano platform. Deactivating, flashing and testing otla ;)

stripwax, 2024-12-16:
My SeeedXIaoM0 is running fine and is also not accelerated with SoftI2CMaster, that's why I mentioned that earlier  :-) 
STM32 and XiaoM0 both run a lot faster than nano, so it's really not easy to estimate different performance behaviour

stripwax, 2024-12-16:
ok @lotharek1977 based on what @rcmolina confirmed, I think you should just try using the prebuilt firmware and see if it works for you.  If it does then the problem is coming somehow from the builds you are doing - and there are lot of variables there around why that might happen - and you might be better off using PlatformIO instead of Arduino IDE

stripwax, 2024-12-18:
Which flavour of stm32f103 do you have?  I want to make sure we are able to build the right variant

lotharek1977, 2024-12-18:
Hello,

generic stm32f103C8 running at standart 8mhz

stripwax, 2024-12-18:
Ok.  Hm, going back to your screenshots right at the top, left hand side was "Maple Mini", right hand side was "Maple Rev 3", so you can see where my confusion was coming from. And when you say "8mhz", that's the crystal but your CPU is running at 72Mhz yes?  Here are the options:
[image]

I think that's sufficient detail but now I'd like to compare to what Rafa's device is using.

@lotharek1977 did you try with one of our downloadable files?  You should just be able to upload it using the programmer, without having to compile anything.  Click here and expand on Assets to see what we provide for all the different platforms.
https://github.com/rcmolina/MaxDuino/releases

lotharek1977, 2024-12-19:
Hello
did try of course. 2 files present in the assets. none of them works for me
diffrentc pinout probably

stripwax, 2024-12-19:
Should be the same pinout as it's built from the same source code with the
same config files. One with 128x32 oled and one with 16x2 LCD.

Rafa - do those STM32 release firmware builds work for you? If not this
might be teething trouble on the platformio build for those devices but
then we should fix that.

lotharek1977, 2024-12-19:
both files cf0/no uploaded via st link gives "blank screen".. seriously :-)

just downloaded /again/ fresh arduino, sd fat, firmware from GITGUB to compile and uploaded. device works. defaults everywhere

i am afraid some diffrence exists :-)

rcmolina, 2024-12-19:
@stripwax ,  my hardware selected before ide building is Mapple mini at 72 Mhz. I will check the best way to flash this, because my board is not an standard maple mini, So only have usb access. Not sure if I can can flash the hex with the same window app I use for nano.

Anyway, I suspect @lotharek have changed buttons and 8 MHz is really strange for me. When implementing timers, not sure if this clock speed was expected. I'll make more investigation forcserial flashing with my stm32 ”maple mini"

lotharek1977, 2024-12-19:
https://lotharek.DSCloud.biz:5001/sharing/dfQwlsWVq

rcmolina, 2024-12-19:
@stripwax , this is what my build shows:
Las variables Globales usan 5600 bytes (32%) de la memoria dinámica, dejando 11808 bytes para las variables locales. El máximo es 17408 bytes.
C:\Users\Rafael\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2020.8.9/win/maple_upload.bat COM4 1 1EAF:0003 C:\Users\Rafael\AppData\Local\Temp\arduino_build_785855/MaxDuino.ino.bin 
maple_loader v0.1
Resetting to bootloader via DTR pulse
[image]
---mappñe_upload.bat---
@echo off
rem: Note %~dp0 get path of this batch file
rem: Need to change drive if My Documents is on a drive other than C:
set driverLetter=%~dp0
set driverLetter=%driverLetter:~0,2%
%driverLetter%
cd %~dp0
java -jar maple_loader.jar %1 %2 %3 %4 %5 %6 %7 %8 %9

for /l %%x in (1, 1, 40) do (
  ping -w 50 -n 1 192.0.2.1 > nul
  mode %1 > nul
  if ERRORLEVEL 0 goto comPortFound
)

echo timeout waiting for %1 serial

:comPortFound

rcmolina, 2024-12-19:
I think the behaviour is very similar to IDE file binary, I was not testing to right back end (spectrum). Sorry, another test bed :)

rcmolina, 2024-12-19:
Another comment on play/stop. The additional pause at the end of the file, was removed in the latest development. So if the last block of a tzx file has 0ms pause, file playing hangs and you need to stop manually the file, changing this value to 100ms solves "the hanging".

rcmolina, 2024-12-19:
For sure, this is not specific for STM32 the same on nano). I'll send you an ID15 file with no ending pause in a minute.

stripwax, 2024-12-20:
Fix for that particular file-ending issue is now available: https://github.com/rcmolina/MaxDuino/pull/75

But from the original report, there still remains some other peculiar issue: "sound comes out of it. what is odd... despite loading sound is hearable, zx spectrum doesnt react wothoiut previous stop being pressed"
But hopefully this change also fixes that too and then we can close this, even if we cannot ourselves reproduce this second problem. 

I'll create a separate ticket around standardising our platformio STM32 builds and making sure that those work on the known devices.

stripwax, 2024-12-20:
@lotharek1977 the comment about "8Mhz being strange" is related to my earlier screenshot: please see the choices available in IDE.  We were still trying to get confirmation on your build settings for your device.
[image]

lotharek1977, 2024-12-30:
Hello, i have attached all informations/anwers in video recorded.

https://lotharek.DSCloud.biz:5001/sharing/hEHzJUGjJ

stripwax, 2024-12-30:
Great video, thank you. So to confirm, the problem seems to be nearly the
same in the old version and the new version. In the old version, it gets
into a state where output is somehow incorrect unless stop is pressed; in
the new version it gets into this same state sooner. It's probably the
timer code not working correctly for your device. I think someone who is an
expert in STM32 might need to investigate.

stripwax, 2026-03-05:
@lotharek1977 Did you ever solve your problems by yourself?  What I would still be keen to do is have a configuration for your device, as part of the platformio.ini build configurations, even if we have no way of testing it.  Or, if you could send me or Rafa one of your devices, we could test and validate that the build is working correctly.  (Or fix it, if it is not working correctly).  Otherwise you are missing out on lots of latest features and improvements.

Those old download links (e.g. your videos) do not work any more, unfortunately.

@rcmolina we should also test and validate the STM32 Maple MiniDuino device that you have and include those settings again properly too, even for OTLA.  Currently none of the STM32-based devices are automatically built.  I had a look at Antonio's website but it doesn't seem like the STM32 devices are for sale, only the ATMEGA32u4 ?  So I would be keen to understand how to test any of these STM32 devices.

stripwax, 2026-03-09:
Ok done - I've created a new config for the device and re-enabled the ATMEGA32U4_Miniduino automatic builds.  The config file is userATMEGA32U4.h  .   That should all be added to the master repo soon I hope.  I've also included all your docs plus a couple of small changes from me.  Thanks again for so much help here!

(TODO for me: I want to separate 'hardware configuration' from 'user configuration', to make it much more obvious what settings you should leave alone for the firmware to actually work in a given device).

stripwax, 2026-03-14:
> @stripwax <https://github.com/stripwax>, I tried but only supported in
> HAL core :(

We should probably try building with the latest official STM32 supported
core instead of the old maple stuff. I did have a branch for it which I can
resurrect, would you be able to test?

rcmolina, 2026-03-14:
Yes of course, no problem. I'll do that for you.

stripwax, 2026-03-16:
@rcmolina  Updated branch here: https://github.com/stripwax/MaxDuino/tree/stm32_official_new
Only a small number of changes required to complete the build, but I have no way of knowing if it works :-)  This uses the official stm32 core package instead of the outdated maple core package.  The timer code is vastly simplified, basically just an interface class and one line of initialisation now.  And this firmware build is also quite a bit smaller than the maple build.

Very interested to know if this works at all, and if so whether it addresses the performance issues seen on OTLA