rcmolina/MaxDuino #66: STM32 MAXDUINO build

lotharek1977, 2024-10-25:
Hello,
as You are laready aware, i have designed small stm32 based maxduino compatible device.

link is here: https://lotharek.pl/productdetail.php?id=409

stm32 build works very good in general..
arduino IDE 1.9.xx (!!) and usb /maple programming works ok

it would make me happy /and probably my future customers/  if you accept proto of my device to develop further firmware with stm32 version.  All i can offer is any device of my offer and happines of mine and my customers worldwide :-)

as usual, please drop me mail :-) or close this issue :-)

rcmolina, 2024-10-25:
Sorry @lotharek1977 ,  I do not accept new prototypes. I have one working from Antonio Villena, and a double board (accets 2 distinct chips) from MerlinkV.

Anyway I appreciate your interest.

Rafa

closed 2024-10-25 by rcmolina

stripwax, 2026-03-28:
@lotharek1977  feel free to send one to me, however, even if Rafa is not interested.  Sorry, I never responded back to you on this.