rcmolina/MaxDuino #76: Ensure platformio STM32 builds are useable/useful on known devices (or change/add configurations)

stripwax, 2024-12-20:


stripwax, 2024-12-20:
Note: there are (at least) two STM32 board packages for Arduino IDE, and more than one platformio core also.
In Arduino IDE:
[image]
You could use the latest version of the "Arduino_STM32" or you can use the package from the stm32duino github page, which seems to be much newer and better supported (but different) https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/refs/heads/main/package_stmicroelectronics_index.json

Currently the MaxDuino only compiles with the Arduino_STM32 in Arduino IDE, and not stm32duino.
With stm32duino, because the build flags are different, various submodules don't compile (e.g. TimerCounter.cpp).
PlatformIO also has support for stm32duino.  So maybe we could move to that, and update the MaxDuino code to build under that environent.
PlatformIO also has a maple build core, which seems to be mostly similar to Arduino_STM32, but resulting firmware is apparently less performant, and definitely comes out larger size than Arduino IDE.  Needs further testing.

We should ensure that both the userSTM32config(s) and the platformio.ini definitions for STM32 match the known devices in circulation, so that the automatic builds are actually useful.

Lastly we should check and confirm that there is a way to upload the automatic builds onto device, and add it to the README.md docs

stripwax, 2025-01-02:
I was having a look at the STM32 TimerCounter code and it's really weird - it's using the same #define TIMER_CHANNEL to refer to both the timer INSTANCE and the timer CHANNEL.  Also, it doesn't build using the official stm32duino core libraries, only using the (outdated? unsupported?) maple core libraries.
I'd suggest modifying this code to build on the supported stm32duino core libraries, and I imagine we'll have a more successful solution that way (for both Arduino IDE and PlatformIO).

I have no way of testing it, but I created this branch, which builds ok using both platformio (I modified env STM32_genericSTM32F103C8_CF0 to be the stm32duino build and the regular STM32_genericSTM32F103C8 is the maple build), as well as Arduino IDE (select the "STM32 MCU Based Boards" from board manager - get the packages.json according to https://github.com/stm32duino/Arduino_Core_STM32 )

With this branch, it won't build correctly using Arduino IDE with the older maple "STM32F1 Boards (Arduino_STM32)" board package  -  we can make that happen too but I'm ignoring that aspect for now.

https://github.com/stripwax/MaxDuino/tree/stm32_official

stripwax, 2025-03-11:
Is the plan now to pretty much disable/abandon STM32?

rcmolina, 2025-03-11:
Of course not, found problems for automatic binary hex generation and commented option in platformio. Not much time now, solving family issues. Back in a few days, not in Madrid now.