rcmolina/MaxDuino #79: OTLA distorted in latest 3.01.1090.96780d

kiFlaFlaki, 2025-03-25:
Have problem with OTLA files distorted in latest 3.01.1090.96780d (Nano 328p) so no use of them.   

in 3.01.1089.9F8F04 OTLA files working great but there are stalling issues with reading SD card, on boot and entering directory with few directories, also when changing from last file/directory to first file/directory.  96780d working really good with SD card but with distorted OTLA files.

rcmolina, 2025-03-25:
@kiFlaFlaki , please be more specific to help you. :)
What configuration are you using? re you flashing the hex in the release ? Which?
Are you using a sample otla files jere? If not, coul you attach it here?
I have some optimizations pending to upload, I've just done now. You can try this new changes to check if they improves your issue. Max otla speed working for me with Arduino nano is 17kbps.

Regards,

Rafael

stripwax, 2025-03-25:
Worth understanding if this is an audio (analogue) problem or a software
(digital) one. It used to be the case that 44kHz sample rate works fine
with Nano but it sounds like the latest report is saying that 44kHz no
longer works reliably with Nano maybe? Whereas for kbps that depends more
on quality of audio output amplification and filtering?

kiFlaFlaki, 2025-03-25:
> [@kiFlaFlaki](https://github.com/kiFlaFlaki) , please be more specific to help you. :) What configuration are you using? re you flashing the hex in the release ? Which? Are you using a sample otla [...]

I am using config7 flashed from Arduino IDE. Otla files are test files from here https://github.com/rcmolina/otla_tzx
version that works was from https://github.com/rcmolina/MaxDuino/releases/tag/latest but is changed now.
I will try to flash it that new version.

kiFlaFlaki, 2025-03-25:
OK, I updated firmware from Arduino IDE from here https://github.com/rcmolina/MaxDuino/tree/master/MaxDuino
I flashed userconfig7. In menu version info I get 3.01.10.8db699. Working with SD card is fast and ok, no stalling. But OTLA playback from here https://github.com/rcmolina/otla_tzx now pilot sound is distorted. 
I than reflashed only .hex [from here https://github.com/rcmolina/MaxDuino/releases/tag/latest](https://github.com/rcmolina/MaxDuino/releases/download/latest/Nano328p_CF7_firmware_Nano328p_CF7.hex)
I get Menu info version 3.01.1090.96780d (this is now latest version from main post) and now OTLA loading works without problems, but working with SD card is slow as in main post. (there is a change as it Maxduino now boots very fast).

rcmolina, 2025-03-25:
@kiFlaFlaki , David is sd expertise, but my opinion based on experience:

- check and report your sdfat library version. 
- What versions are you using?
- When  building with IDE I use old versions 1.4 and 2.0.
- My experience is version 2 works very well with claas 10 sds. . What sd class and capacity are you using? Can you test with different ones?
Thanks a lot. For sure David will help us.

Rafa

kiFlaFlaki, 2025-03-25:
How can I check current sdfat library version? In library manager I get displayed latest version and install option. 
My SD Card is Kingston Canvas Select Plus SDHC v10 32GB, tried another brand same issues.

rcmolina, 2025-03-25:
@kiFlaFlaki , look here:

C:\Users\Rafael\Documents\Arduino\libraries
You can add an underscore at the beggining of your SdFat directory. Look inside library properties:

name=SdFat
version=1.1.0
license=MIT
author=Bill Greiman <fat16lib@sbcglobal.net>
maintainer=Bill Greiman <fat16lib@sbcglobal.net>
sentence=FAT16/FAT32 file system for SD cards.
paragraph=FAT16/FAT32 file system for SD cards.
category=Data Storage
url=https://github.com/greiman/SdFat
repository=https://github.com/greiman/SdFat.git
architectures=*

When building with my IDE 1.8.4 I use 1.1.0 and some times 2.2.2
Check here and copy this version to test and build again:

https://github.com/rcmolina/MaxDuino_BETA/tree/master/Troubleshooting/libraries/SdFat

Please report versions for all of your libraries :)

Rafa

kiFlaFlaki, 2025-03-25:
I use 1.8.10 Arduino IDE
Testing version on https://github.com/rcmolina/MaxDuino_BETA/tree/master/Troubleshooting/libraries/SdFat is exactly the same that I use. So I can try with suggested 1.1.0 or 2.2.2?
Thanks for helping,

Damir

rcmolina, 2025-03-25:
@kiFlaFlaki , my SD is 8GB, if cant use a 2 GB or 8GB SD card for testing, I recommend you to use 2.2.2 version:
https://github.com/greiman/SdFat/archive/refs/tags/2.2.2.zip

kiFlaFlaki, 2025-03-25:
I installed 1.1.0 version and build with that.  Now everything works as it should.  No problems with OTLA files, they work fantastic, 17kbps also, and SD Card works fast.  Strange because I did not have problems before.  Did not tried 2.2.2 for now. 
Thank you very much for help and time Rafa! i flashed now both my maxduino’s and they works great now.

closed 2025-03-25 by rcmolina