rcmolina/MaxDuino #80: Last block of TZX files does not finish properly (some TZX files only)

TomCat666, 2025-03-25:
Hi

Noticed that on some TZX files the last block is not fully finished - it seems the (at least) the last wave is not properly done.
Here is a test TZX file that fails loading on ZX Spectrum - just do LOAD "" CODE and you will get R Tape Loading Error.

This is using NANO v3 OldBoot Maxduino v3 firmware.

[test.zip](https://github.com/user-attachments/files/19456726/test.zip)

rcmolina, 2025-03-25:
@TomCat666 , hi Tomaz , my respects, you are a pioneer for all of us, lernes a lot from your code.
I've tested your submitted tzx file, a very short code. No problem for me, I can save a video if you prefer this as a proof.
This kind of problems seem to be related to sd speed type or sdfat library version (I'm using IDE 1.8.4 and sdfat 1.1.0).
Attached find my hex from IDE, you can try to upload with XLoader and check if this solves the issue. Also changing to smaller SD apacity card can be checked.

[MaxDuino.OLED.096inch,nanoCF7.zip](https://github.com/user-attachments/files/19457033/MaxDuino.OLED.096inch.nanoCF7.zip)

Check also this issue, it can help you also: https://github.com/rcmolina/MaxDuino/issues/79

TomCat666, 2025-03-25:
Hi rcmolina. Was fun working on TZX and stuff, but that code was BAD :D I am glad it was usefull though.

Tried a smaller/older SD card and the same result.

When uploading your hex file the display is not working - I have the 2x16 chars display on 0x27 address... probably yours is compiled for some other display?

rcmolina, 2025-03-26:
@TomCat666 , do not agree, you are a genious, also on C64 computers :) I still use your z802tzx3 modified here, previous compression (zx7 or zx0):  https://github.com/rcmolina/z802tzx3
Regarding your questio, sorry I suppose you have an oled version screen :( So, for an LCD 0x27 you need userconf3 params. Find attached here again theworking hex manually compiled from IDE 1.8.4 and sdfat 1.1.0: 

[MaxDuino.ino.eightanaloginputs.zip](https://github.com/user-attachments/files/19461235/MaxDuino.ino.eightanaloginputs.zip)

There is also an automatic building in the project with different library versions, test it if you want:
https://github.com/rcmolina/MaxDuino/releases/download/latest/Nano328p_CF3_firmware_Nano328p_CF3.hex

Rafa

kiFlaFlaki, 2025-03-26:
Loading test.tzx  here also get R Tape Loading Error when  TSXCzxpUEFSW is ON. When TSXCzxpUEFSW is OFF it loads OK. Is that suppose to be, as TSXCzxpUEFSW ON is default?

rcmolina, 2025-03-26:
@kiFlaFlaki , thanks. Normal position is off. This TSXCzxpUEFSW is a multipurpose button, I'll try to sumarize functions:

- for zx spectrum, polarity. Forces inversion low-high with pauses, allowingbloadingbgames like Basil the great mouse detective
- for msx is tsx control. You can change and force the speed por tsx file
-  for amstrad cdt, you can force speed up for id11 blocks
-  for bbc micro, you can switch parity odd-even to deal with a preservation tool problem.

Waiting first for Tomaz feedback, I'll test this later.

Rafa

kiFlaFlaki, 2025-03-26:
Thanks for clarification Rafa! Indeed there are Spectrum titles that won't load with TSXCzxpUEFSW ON position and vice versa.
MsPacman.tzx for example, loads in ON position.

rcmolina, 2025-03-26:
@TomCat666 , verified the issue. Some changes ruined the code :) After some backward testing, this was the last version working for me: 
https://github.com/rcmolina/MaxDuino_BETA/tree/master/Troubleshooting/20241016_1/MaxDuino

In the meanwile, attached find the hex for userconfig3, so please check if also works ok for you:

[MaxDuino.LCD.0x27.zip](https://github.com/user-attachments/files/19472189/MaxDuino.LCD.0x27.zip)

Thanks a lot

Rafa

rcmolina, 2025-03-26:
@TomCat666 , fixed
Check new version, confirm this to close the issue.

Rafa

TomCat666, 2025-03-26:
Will do as soon as I am at my computer. :)

TomCat666, 2025-03-26:
Yes, confirming it is fixed with the latest fw you provided. Thank you! :)

closed 2025-03-26 by rcmolina

rcmolina, 2025-03-27:
last release patched for LCD 0x27 (with jupiter ace tap support):

[MaxDuino.LCD_0x27.250311.zip](https://github.com/user-attachments/files/19490703/MaxDuino.LCD_0x27.250311.zip)

TomCat666, 2025-03-27:
can't test jupiter ace, but seems to be working fine otherwise :)