rcmolina/MaxDuino #83: Idea: verify (test) the ESP builds

stripwax, 2025-06-25:


Hamberthm, 2025-08-07:
> All the sourcecode is in the git repository.... The release binaries are
> generated automatically from the sourcecode....
> 
> But please note that the ESP binaries are essentially untested, and that
> the MaxDuino software is not yet really validated on ESP : I have one that
> I built myself, and I coded up the firmware changes to make it work but
> it's not loading very reliably for me. Something odd about timings that
> nobody finished investigating.
> 
> Also I'm fairly sure the binary is for C3 not S3 (I could be wrong)
> […](#)
> 
>  <[#83 (comment)](https://github.com/rcmolina/MaxDuino/issues/83#issuecomment-3121993843)>
> 
>  Hi! I'm new on the project. We successfully used the project at our retro
>  tech museum, Espacio Tec <http://www.espaciotec.com.ar>. Thanks for the
>  work!
> 
>  Now I just bought these Waveshare ESP32-C6 development boards
>  <https://www.waveshare.com/esp32-c6-touch-lcd-1.47.htm>, I plan to
>  develop a new UI for the project and again, use it at the museum.
> 
>  I see there's compiled binaries at the releases page for the ESP32-S3,
>  which as far as I understand, it's also a RISC-V processor.
> 
>  I was wondering if I can get the configured source files for the S3, as I
>  can see the binaries but not the sources. Thanks!!
> 
>  <[#83 (comment)](https://github.com/rcmolina/MaxDuino/issues/83#issuecomment-3121993843)>,

Hi again! You are right, the processor on this module is C6 variant.

I've compiled and adapted the project to the Waveshare module, and made a rudimentary UI.

I've reached the point of testing the playing process, but sadly it is not working.

As far as I debugged, it seems the ISR function wave2() is not being called by the interrupt service at all.

I'll continue to debug and come back with my findings. Any hand on this will be of great help.

Hamberthm, 2025-08-08:
Found the problem.

For ESP32 boards, the timer is attached to the `ESPTimerCallback` variable for the callback function, as defined here:

https://github.com/rcmolina/MaxDuino/blob/9fe44db4be39b576c434f15184d106db90bf8707/MaxDuino/TimerCounter.cpp#L558

But the problem is, in the `attachInterrupt()` function, the cb functions gets assigned to an unused `isrCallback` variable in:

https://github.com/rcmolina/MaxDuino/blob/9fe44db4be39b576c434f15184d106db90bf8707/MaxDuino/TimerCounter.cpp#L593

The solution is to modify this line to:

` ESPTimerCallback = isr; `

I will upload a fork working for the Waveshare module as soon as I finish polishing it!

stripwax, 2026-03-03:
@Hamberthm - were you able to confirm your fix works for your device?  And did you create a fork and pull request?

stripwax, 2026-03-28:
@Hamberthm  oh - one other request please - you say that you adapted MaxDuino for the Waveshare board - can I ask a few things please -.   1 . what did you need to change and  2.  what was the advantage/reason for using ESP32-C6 rather than ESP32-C3 ?   

Ideally we can support all of this in the main repository so if you already know what to change and can let us know, we'll add it.  (But we won't be able to test it ourselves without such a device)